MrYsLab/s3onegpio #1: OnePGIO fails to synch when Arduino is disconnected and connected back to USB port.

RoCorbera, 2019-10-09:
If Arduino is connected to USB and Scratch 3 is working on it, OneGPIO fails to synch again when Arduino is plugged again to USB port. It also makes all the ports that were being used when unplugging USB to become instable and unusuable under Scratch.

MrYsLab, 2019-10-09:
Thanks for the feedback. When you say "It also makes all the ports that were being used when unplugging USB to become instable and unusuable under Scratch.", do you have other serial devices plugged into the USB ports? If so, what type of devices are these?

There is a known problem with Firmata in syncing resets. Since the extension is intended for use with the Arduino Uno, I do have a way to force a hardware reset of the Arduino. I need to try that feature out and see if it helps for the extension. Unfortunately, the auto hardware reset does not work on all Arduinos because of differences in the processor architectures.

RoCorbera, 2019-10-11:
No, I just had one Arduino board plugged to USB. This issue also happens when I hit reset button on Arduino. For some reason the pin I was using in a Scratch 3 program become unaresponsive after Arduino reset or having USB conected and reconected again.

MrYsLab, 2019-10-13:
I spend a little time looking at this. Using the following scenario:
1. Scratch 3 and extension for Arduino are up and running.
2. Kill the Python extension code.
3. Pull the USB plug on the Arduino and then plug it in again.

Using my yet to be released Python extension code, on Ubuntu 19.04, I see a similar problem. It appears that even though I killed the Python process, a zombie process is still running. If I wait for about 10 or 15 seconds before trying to restart the python extension, then things seem to behave as expected.

I tried the same using a Raspberry Pi with the latest Buster releases of Raspbian, and I can restart the extension without having to wait. 

I have not tested yet on Windows, so I am not sure if a wait might be needed for it as well.

For Mac, unfortunately, I do not own one so I do not know its behavior.

I will document this when I publish the extension release.

Thanks for your input.

MrYsLab, 2019-10-17:
I am closing this issue with the official release.

closed 2019-10-17 by MrYsLab