MrYsLab/s3onegpio #2: Read Digital and Read Analog don't work

RoCorbera, 2019-10-09:
If Scratch tries to run any sort of Read, OneGPIO crashes and everything else stops working.

MrYsLab, 2019-10-09:
The OneGPIO code located in this repository will be removed in the very near future. Newer versions will be made available and are used to launch the Banyan components that make up the Python extension code for each board.

The code and all the installation and usage documentation will be located at:
https://github.com/MrYsLab/s3-extend.

The new extensions have been tested and digital and analog inputs are functional. I hope to have the documentation finished before Oct 18, 2019, and will release everything then.

BTW, if you have any of the OneGPIO blocks being used in a loop, a short wait block of .001 seconds needs to be added to the loop. Also, sometimes it is necessary to have the reporter blocks in their own Scratch thread, and placing the value of the read into a variable that can be used by other parts of the Scratch script.

I will let you know when s3-extend becomes available.

RoCorbera, 2019-10-11:
Thanks for clarifying it. I'll wait for the new release.

MrYsLab, 2019-10-17:
I am closing this issue with the official release.

closed 2019-10-17 by MrYsLab