MrYsLab/s3onegpio #3: Utrasonic doesn't work

RoCorbera, 2019-10-09:
When trying to read ultrasonic sensor, Arduino board starts to receive commands from OneGPIO but never stops. This leads T/E ports to become unusable under Scratch and maybe a non-stop data sending from PC to Arduino.

MrYsLab, 2019-10-09:
Please see my answer to issue #2. Data changes on any input, either Analog, or Digital will always be sent to Scratch 3, so depending upon the sensor, the data may be non-stop. If this overwhelms the Scratch 3 code, I will investigate a way to throttle the incoming data.

RoCorbera, 2019-10-11:
In my testing the issue is related to blocking selected PINs (T/E) from being used again. Given it's not possible to stop the usltrasonic reading process upon started, those pins are blocked for any other use. I'm not sure if this issue may be related to the issue #2.

MrYsLab, 2019-10-12:
When you say you the pins are blocked for any other use, are you intending to remove the HC-SR04 and repurpose the T and  E pins while the Arduino is powered up?
Could you please explain why and how you would "reuse" the T and E pins once the Sonar block is executed? Thanks.

RoCorbera, 2019-10-13:
Well, I tested it with a kid playing with Scrath! This kid decided to unplug the SR04 and by this I found out this issue.

In summary: 

1- If the kid uses a block (by accidentally or just testing by clicking on the Scratch3 block) and there is no SR04 connected to the T/E pins, they can not be used anymore given the will receive input for ever. 

2- When using SR04 with correct setup for pins T/E and the kid decides to unplug it and create a new project.... those pins get for ever receiving input and can't be used leading the kid to think it is broken.

MrYsLab, 2019-10-13:
Yes, kids will be kids - I ran a Scratch club for 9-13-year-olds, so I understand. It can be bad for the health of an Arduino if items are plugged in and out while the Arduino has power applied. Since this is an "operational" error, I am not likely to address it since it is not a bug.

What I would suggest is that when one is finished using sensors and actuators, to first save the Scratch script if one wishes to run it again in the future. Next stop the Banyan code that connects Scratch 3 to the board (in the upcoming release there will be a single line command to start the "connector". To kill it, one would hit Control-C) and lastly power down the Arduino, rewire it and then repower the Arduino.

MrYsLab, 2019-10-17:
I am closing this issue with the official release.

closed 2019-10-17 by MrYsLab