MrYsLab/s3onegpio #4: WebSocket Connection Is Closed - a way to reconnect w/o refresh browser

HackEduca, 2019-10-18:
Hi Alan, I am doing some test with PWM and sometimes receive a message **"WebSocket Connection Is Closed"** To leave this need to refresh the browser.

Is there a way to  reestablish the connection without refresh the browser - once we loose what we are doing?
Adding a button like you placed in previous version?
Or any other trick...

MrYsLab, 2019-10-18:
Hi Edson,
    I recreated your Scratch script and did not see the WebSocket closing. I also never saw the value of 'a' go above 1 - I am not sure why that was, so I changed the script to this:
[image]

If you notice I added a delay after writing to the LED. You may try adding a small delay to see if the WebSocket close goes away.

I am not sure why you could not just click the green button again. When the WebSocket closes, if you look at the 2 windows on your pc that were created when you ran s2a, do either of these show an exception?

If not, if you could provide the steps I would use to generate the WebSocket close that would help me find the answer.

Thanks,
Alan

HackEduca, 2019-10-19:
About the code it was just a brainstorming - not complete.
Looks that the delay (as mentioned in your tutorial), worked this time.

I did another code and I tried to get the same error, but not always.
[Project_PWM_2.zip](https://github.com/MrYsLab/s3onegpio/files/3747049/Project_PWM_2.zip)

To recreate te error I just removed the waiting block when program is in execution.

[WebSocket_Closed.zip](https://github.com/MrYsLab/s3onegpio/files/3747058/WebSocket_Closed.zip)

You can see that there is no error message in the prompt.
[image]

In order to work again, just refreshing the web browser.

Thanks,
Edson

MrYsLab, 2019-10-20:
Thanks. I ran your program for 2 or 3 minutes and also started, stopped and restarted several times and do not see the WebSocket closing. Are you still seeing it closing when the delays are there?

If you are still seeing the close alert, is there a list of steps I can follow to reproduce the error?

Thanks, 
Alan

HackEduca, 2019-10-20:
Hi Alan, did you take a chance to see the attached file (WebSocket_Closed.zip)?
I will send you a new video (in a zip file)
In that video I tried  simulaton to close the Websocket.

- Please to open the following Scratch file:
[Project_PWM_5.zip](https://github.com/MrYsLab/s3onegpio/files/3747774/Project_PWM_5.zip)

- Execute and during execution remove those delays block.
This is enough to close the Websocket. From this time on Arduino doesn´t receive any command anymore and back to work again only if browser is reloaded.
I did a test running file on server and local file.
Following a new video file with more details (hope you can see it)
[websocket.zip](https://github.com/MrYsLab/s3onegpio/files/3747776/websocket.zip)

- Alan, probably this is a silly test, but I just want to know if is there another way to be back to code without the browser refresh. Is not possible to predict what students will do on class.

Thanks
Edson

MrYsLab, 2019-10-20:
Edson,
    Thanks for taking the time to allow me to reproduce the problem. What I wanted to know was if the browser or the python code closed the socket. It is on the browser side, so without the wait, the WebSocket client built into the browser is getting overwhelmed with send requests. 

What I am thinking of doing is to add the delay in the javascript block code so that the user doesn't need to use the wait blocks. I am not sure if this will work because with javascript it is not as simple as putting in a sleep call. I will also check on the web if there is an easier way to solve this problem.

Give me a few days and I will let you know if I was able to solve this.

Thanks,
Alan

MrYsLab, 2019-10-20:
I was able to add a delay to the pwm block by using [lodash throttle](https://lodash.com/docs/4.17.15#throttle). I tried various delays from 1 to 100 milliseconds and the WebSocket still closed. 

I then modified your example. I changed the pin number from 5 to 6 to match my hardware - this has nothing to do with the issue. I then moved each of the sets of blocks to be in its own sprite.
I removed all 3 of the 1 ms. delays. It has run for over an hour without a crash. I tried twice. So, I think the problem is in the Scratch code when running multiple block threads in a single sprite. 

I am attaching the project so you can take a look at what I did.

Thanks,
Alan
[Project_PWM_6a.zip](https://github.com/MrYsLab/s3onegpio/files/3748168/Project_PWM_6a.zip)

HackEduca, 2019-10-21:
Thanks to try to get a prompt solution.
This is what I got from them - I think is incomplete
[image]
------
By the way this is not fair.... Scratch 3.0 still on development and I think will not be easy to identify the root cause for all future issues.
Thanks again
Edson

MrYsLab, 2019-10-21:
This is going to be difficult to find and correct. The Scratch team is unlikely to run my extension to hunt down the problem and since there is no trail left in the browser console, there is not much I can do. I had seen similar problems in Scratch 2, so I am not surprised. I am going to close this unless you think it should be open. I can see comment here after I close the issue.

closed 2019-10-21 by MrYsLab