MrYsLab/s3onegpio #15: How to start the OneGPIO server python code?

doutv, 2020-01-10:
I've come across this problem when I'm trying to use Arduino Nano. I use Windows and can't just type `s3a` to run the python code. Directly download your s3-extend code and run s3a.py also fail.

MrYsLab, 2020-01-10:
The Uno is the only officially supported board, but I am willing to look at your issue. Please[ follow these troubleshooting instructions ](https://mryslab.github.io/s3-extend/#ext_ard/#troubleshooting) and let me know if there are any exceptions reported on any window and what you see on the ardgw window.

MrYsLab, 2020-01-10:
Also, did you load FirmataExpress on the Nano?

doutv, 2020-01-11:
I have load FirmataExpress on Nano.
Actually, I don't know how to start [s3-extend](https://mryslab.github.io/s3-extend/#ext_ard/#troubleshooting) on my computer. I download your [code](https://github.com/MrYsLab/s3-extend) and run s3a.py. Is it the correct way to start python server?
```
Traceback (most recent call last):
  File "c:/Users/Jason/Documents/GitHub/s3-extend/s3_extend/s3a.py", line 251, in <module>
    s3ax()
  File "c:/Users/Jason/Documents/GitHub/s3-extend/s3_extend/s3a.py", line 242, in s3ax
    S3A(com_port=com_port, arduino_instance_id=args.arduino_instance_id)
  File "c:/Users/Jason/Documents/GitHub/s3-extend/s3_extend/s3a.py", line 57, in __init__
    self.proc_bp = self.start_backplane()
  File "c:/Users/Jason/Documents/GitHub/s3-extend/s3_extend/s3a.py", line 174, in start_backplane
    subprocess.CREATE_NO_WINDOW)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] 系统找不到指定的文件。
```
Following your tutorial, I open cmd on Win10 , enter the command s3a and it says that it isn't a program or command. Should I add [s3-extend](https://mryslab.github.io/s3-extend/#ext_ard/#troubleshooting) file to my system path?
> Step 3: Start the OneGPIO server code by opening a terminal window for each microcontroller you wish to use and then enter the command for the specific micro-controller:
> 
> s3a - for the Arduino
> s3c - for the Adafruit Circuit Playground Express
> s3e - for the ESP-8266
> s3p - for the Picoboard
> s3r - for the Raspberry Pi

Thanks for your help, your project is great when I'm looking for a scratch+arduino platform.

MrYsLab, 2020-01-11:
You should not need to change your system path. To make sure that Python was installed on the path,
open a command window and type:

```
python
```

If you see a banner that shows python 3 is running you can close Python and should be able to do the following:
1. In a command window type:

```
pip install s3-extend
```

This will install s3-extend and all of the packages it uses. It will also install s3a as an executable. So to start the extension for Arduino, in a command window enter:

```
s3a
```
Please let me know if you are still having an issue, and if not, please close this issue.

doutv, 2020-01-12:
Thanks for your help! I've figured out the problem and run it on my Nano successfully.
Below are the problems I came across and their solutions. Maybe it can help others.
# can't find s3a command in cmd
I use `python -m pip install xxx --user` to install python package, but maybe it will cause some problems.
<https://stackoverflow.com/questions/58386953/pip3-typeerror-module-object-is-not-callable-after-update>
So I use `pip install s3-extend` under administrator mode and it works for me.
I think it's because `--user` will install the python package in a different folder like `C:\Users\XXX\AppData\Roaming\Python\Python37\Scripts`
Without `--user` and run in administrator mode:
`C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64`
# subprocess.py error
```
Traceback (most recent call last):
  File "c:/Users/Jason/Documents/GitHub/s3-extend/s3_extend/s3a.py", line 251, in <module>
    s3ax()
  File "c:/Users/Jason/Documents/GitHub/s3-extend/s3_extend/s3a.py", line 242, in s3ax
    S3A(com_port=com_port, arduino_instance_id=args.arduino_instance_id)
  File "c:/Users/Jason/Documents/GitHub/s3-extend/s3_extend/s3a.py", line 57, in __init__
    self.proc_bp = self.start_backplane()
  File "c:/Users/Jason/Documents/GitHub/s3-extend/s3_extend/s3a.py", line 174, in start_backplane
    subprocess.CREATE_NO_WINDOW)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "C:\Program Files (x86)\Microsoft Visual Studio\Shared\Python37_64\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2]
```
<https://blog.csdn.net/u014094184/article/details/80085336>
modify subprocess.py
```python
_init__:
shell=False to shell=True
```
I don't know the reason.
# my environment
Windows10
Again, Thanks for your devotion! I will help you complete this good project.

closed 2020-01-12 by doutv