MrYsLab/s3onegpio #19: Feature - I2C Servo and LED control of PCA9685

dlarue, 2020-02-18:
Do you have this on your radar?  I'd like to start a project elementary age kids which combines a story with various Scratch programming steps to solve problems, create physical control elements and help get an astronaut closer to getting back to Earth. I found I can use Scratch 3 on the default Raspbian to turn on/off LEDs and read switches and voltages but I can not control servo motors or DC motors.  The venerable PCA9685 is in many devices and there's even a rPi hat with 16 channels for ~$10. For ~$20 there's a rPi hat with the PCA9685 and an h-bridge.

Besides the full story, what's missing is the Servo control over I2C.  I did find a board called PivoPi which has Scratch 2 support but I'm shooting for Scratch 3 so I've just started to read "How to Develop Your Own Block for Scratch 3.0".  Thanks.

MrYsLab, 2020-02-18:
Hi Doug. Your project sounds like the kids will really enjoy it. For Scratch extensions, I have avoided using i2c interfaces mainly because the Scratch blocks I created are low level. Although it can be done, I did not think that asking kids to directly program an i2c device using Scratch would be a fun task.

If you are going to create higher-level blocks and if you are comfortable with i2c programming, you can use PCA9685 devices. I did create a generic i2c interface for the [OneGPIO demo stations](https://mryslab.github.io/python_banyan/#one_gpio_demos/#banyan-demonstration-station-web-pages) using the generic OneGPIO commands and reports. These are specified in the [OneGPIO Spec](https://github.com/MrYsLab/python_banyan/blob/master/projects/OneGPIO/message_specification/OneGPIO.pdf). Just look for i2c in the document. 
I have also [implemented Analog input](https://mryslab.github.io/python_banyan/#banyan_gateways/#adding-some-more-spin ) for the RPi using pigpio to control the i2c.

Unless you need to control a large number servos, I think just using vanilla GPIO in conjunction with a library like pigpio is the simplest thing to do. If you are looking for a hat to connect the motors, then this [screw terminal hat](https://smile.amazon.com/gp/product/B01M27459S?pf_rd_p=ab873d20-a0ca-439b-ac45-cd78f07a84d8&pf_rd_r=XYRGMZTBRDJER3ZH8PH2) would do the trick. It even has room to add an H-Bridge for DC motor control.

Since this is not an issue with s3onegpio, I am going to close this.  If you have any questions or comments, feel free to comment here and I will be nofified and reply if necessary.

closed 2020-02-18 by MrYsLab

dlarue, 2020-02-24:
Correct, I don't want kids to program I2C but instead just have a Servo block which can issue I2C commands to the PCA9685 to position servos.

I will look at your example of using I2C in your OneGPIO.  I see now that pigpio does support servo PWM so maybe by looking at your analog example I can get that working for a servo block.

This learning adventure game should only need 2-4 servos at most.

Thanks for your comments and assistance.

MrYsLab, 2020-02-24:
I am not sure what you mean by "servo PWM". I have controlled servos using pigpio pwm using Python. Have you tried the PWM block I have included with the RPi extension? The pin numbers are BCM if that is not clear. If that works, you could create your own blocks within Scratch by clicking on My Blocks and clicking on Make A Block. This would hide all the pin numbers, and pwm details from the kids and would be much simpler than trying to extend what is already there.

dlarue, 2020-02-24:
For "Servo blocks" I was thinking there would be an introduction to angles from 0 to 180 degrees and so the kids would needs to eventually acquire the correct angle to enter into the servo block.  Setting millisecond values is something I wasn't planning on getting into.  

I will evaluate what you have and see how it goes. I was hopefully to start on this by Wednesday with the motor hat as I received the Waveshare Motor Driver HAT this weekend. But I messed up and thought because it has a PCA9685 16 channel PWM driver chip, the hat would have servo outputs. But they only use the PCA9685 to drive the h-bridge speed control. Ugh as it would have cost almost nothing to add 2 or 4 servo connections.  I have an old PCA9685 board around though so still should have enough to investigate.

MrYsLab, 2020-02-25:
Hopefully, this won't send you down a rabbit hole. I created a web-based demo page for the python-banyan framework that sets the angle of the servo for the RPi.  Here is a screenshot:
[image]

The following [code is part of the OneGPIO project](https://github.com/MrYsLab/python_banyan/blob/126f889ac9a83070eef526966771a575c7068745/projects/OneGPIO/raspberry_pi/rpi_gateway.py#L232) and it sets the servo angle. The data comes in as a banyan message. The "payload" contains the pin number and the angle in degrees. If you create a similar function without the Banyan messaging, hopefully, it should work.

```
    def servo_position(self, topic, payload):
        """
        This method will set a servo's position in degrees.
        Call set_mode_servo first to activate the pin for
        servo operation.

        :param topic: message topic
        :param payload: {'command': 'servo_position',
                         "pin": “PIN”,'tag': 'servo',
                        “position”: “POSITION”}
        """
        pin = payload['pin']
        pulse_width = (payload['position'] * 10) + 500
        self.pi.set_servo_pulsewidth(pin, pulse_width)
```

dlarue, 2020-02-25:
Rabbit hole?  I live in rabbit holes LOL.  I'm looking forward to learning more about the Banyan messaging system but will be staying out of the JavaScript side of things for a bit.  And every time I see something new, like Banyan Messaging, I wonder how well it might fit into other projects.
I just had another project reempt my Wed efforts so hope to start on Thur. Thanks!

MrYsLab, 2020-02-25:
I don't blame you for staying away from JS. I break out in a rash every time I need to write some. Also, I didn't mean that you need to learn Banyan at this time. BTW, if you want to see a simple application of Banyan for sending messages from one process to another, check my article on [creating p2p games using arcade](https://mryslab.github.io/bots-in-pieces/python-banyan/arcade/2020/02/21/p2p-arcade-1.html).

Getting back to the topic at hand, here is some simple code for servo control:

```
import pigpio

# remember to:
# sudo pigpiod
# before running

pi = pigpio.pi()
pin = 17
angle = 180
pulse_width = (angle * 10) + 500
pi.set_servo_pulsewidth(pin, pulse_width)
```
If you are going to be moving the servo in a loop to change its position, you will need to
add a delay, since it takes time for the servo motor to move.

Hope this helps.

BTW, I am not sure if you are going to be using s3onegpio or not, but the RPi extension already has a servo block:

[image]

dlarue, 2020-02-25:
Speaking of rabbit holes, do you have any experience with ZeroMQ and if so, have you mentioned anywhere(docs) comparing the two or about your choice of using Banyan.  I once dabbled in CORBA but haven't got back into distributed messaging except for seeing it in ROS. Another project I'm involved in is the donkey platform for autonomous small scale cars(RC) and the primary developer is part of my weekly meetup group. He found the overhead of ROS packaging excessive and opted for ZeroMQ but so far from what I've read on your Banyan Blue Bot Part 4 doc it too looks pretty light weight. At least in the packaging requirements.

MrYsLab, 2020-02-25:
Funny you should ask. Banyan is a wrapper for ZeroMQ configured for pub/sub. It also wraps [MessagePack](https://msgpack.org/index.html)  for data encoding/decoding when going over the network. It supports transporting Numpy data if you should be so inclined. 

To connect things together, all the Banyan pieces connect through something I call the Backplane. It is an instance of a [ZeroMQ Forwarder](https://learning-0mq-with-pyzmq.readthedocs.io/en/latest/pyzmq/devices/forwarder.html). This allows every Banyan component to connect to a common IP address using common IP ports - look ma, no routing ;-).

The Banyan documentation looks a little inhibiting, but it is dead simple to use. The [base class](https://htmlpreview.github.io/?https://raw.githubusercontent.com/MrYsLab/python_banyan/master/html/python_banyan/banyan_base/banyan_base.html) has 6 methods and only one needs to be overridden - that is the method that processes the received messages.

CORBA - a blast from my distant past.