MrYsLab/s3onegpio #20: Translation For French  fr

framboise314, 2020-02-23:
Hello 
here are 2 files with french translation
i will receive very soon the board and will improve translation when doing tests
best regards
françois
[playground_express.txt](https://github.com/MrYsLab/s3onegpio/files/4241471/playground_express.txt)
[picoboard.txt](https://github.com/MrYsLab/s3onegpio/files/4241472/picoboard.txt)

MrYsLab, 2020-02-23:
Thanks. I prefer to wait until you finalize the translations before incorporating them into the code.
If you need to have the current translations incorporated, please let me know.

framboise314, 2020-02-23:
ok thank you
i should receive the boards this week
regards
francois

[image:
https://www.editions-eni.fr/livre/raspberry-pi-4-exploitez-tout-le-potentiel-de-votre-nano-ordinateur-inclus-un-projet-de-station-meteo-9782409022081]
<https://www.editions-eni.fr/livre/raspberry-pi-4-exploitez-tout-le-potentiel-de-votre-nano-ordinateur-inclus-un-projet-de-station-meteo-9782409022081?utm_source=Framboise314&utm_medium=affiliation&utm_campaign=20200109-Mocq&xtor=AL-3908-%5bFMOCQ>
https://www.framboise314.fr
https://www.editions-eni.fr/francois-mocq

framboise314, 2020-03-19:
Hello Alan 
here are the 2 files with french translation 
i just made slight modifications
I played with Playgound Express / PicoBoard / ESP8266 and Arduino
some issues with Playground and PicoBoard 
[PicoBoard Pi4 Scratch 3_EN.pdf](https://github.com/MrYsLab/s3onegpio/files/4355476/PicoBoard.Pi4.Scratch.3_EN.pdf)
you'll find screen copy of the tests here : s3p don not close but other commands return errors 
and when i move the slider, the two led Rx & Tx became off
i go on with Playground Express will send a report later

[picoboard.txt](https://github.com/MrYsLab/s3onegpio/files/4355456/picoboard.txt)
[playground_express.txt](https://github.com/MrYsLab/s3onegpio/files/4355458/playground_express.txt)

MrYsLab, 2020-03-19:
Hi Francois,
    I will add your translations within a day or two. Thank you.
For the problems you are seeing are you starting the program using s3p or s3c and then starting the extension in the browser, or are you also starting the backplane and gateways after starting s3p or s3c?
You just need to start s3p or s3c all the gateways will be started for you, you do not need to start anything else.

I tested with the CircuitPlayground on a new installation of Raspbian on a Raspberry Pi 4 and I cannot reproduce the errors. If you provide the steps used to create the errors, I will try to recreate them here.

For the Picoboard, this may sound strange, but I do not own one. I developed the Picoboard extension at the request of an educational group in Taiwan and they did the testing for me. The Picoboard they are using is a clone created in Taiwan. If you are using a Sparkfun (which they did not test) may behave differently. I test my code using a Picoboard simulator that I load into an Arduino. 

Again, let me know the exact steps you are taking to create the problem.

Thanks once again. Stay well.

Alan

MrYsLab, 2020-03-20:
Hi Francois,
    I have added your translations and they are now available online. Thank you.
I am going to close this issue. If you are still having issues with s3-extend, could you please
enter a new issue [here.](https://github.com/MrYsLab/s3-extend/issues)
Thanks

closed 2020-03-20 by MrYsLab

framboise314, 2020-03-21:
Thank you Alan
have a nice day, take care