MrYsLab/s3onegpio #21: 一但使用pwm的模块就网络中断。。。

lsyaizyl, 2020-05-02:


lsyaizyl, 2020-05-02:
具体如何解决，大神？

lsyaizyl, 2020-05-02:
我是使用scratch离线版的

MrYsLab, 2020-05-02:
I don't know if you mean the offline version provided by the Scratch team or my version. You must use my version. You can [find instructions here.](https://mryslab.github.io/s3-extend/#s3-local/)
If you are using my version, what browser are you using (Chrome, Firefox, etc)?

lsyaizyl, 2020-05-02:
我使用在您这里下载的版本，Chrome

MrYsLab, 2020-05-02:
Ok. You will need to follow the troubleshooting instructions as mentioned in issue #22.

lsyaizyl, 2020-05-02:
pi@raspberrypi:~ $ s3r
Only run this script on a Raspberry Pi!
backplane started
Websocket Gateway started
RPi Gateway started 
To exit this program, press Control-c
RPi Gateway exited.
pi@raspberrypi:~ $

lsyaizyl, 2020-05-02:
[image]

lsyaizyl, 2020-05-02:
[image]

lsyaizyl, 2020-05-02:
[image]

MrYsLab, 2020-05-02:
Now when you run s3onegpio and try to run PWM, does it crash? If so, is there any exception output in any of the screens?

lsyaizyl, 2020-05-02:
[image]

lsyaizyl, 2020-05-02:
不起作用了

lsyaizyl, 2020-05-02:
[image]

lsyaizyl, 2020-05-02:
已杀死

lsyaizyl, 2020-05-02:
[image]

MrYsLab, 2020-05-02:
What pin number are you using in the Scratch block and what is percentage value you are using?

lsyaizyl, 2020-05-02:
无论是多少都会出现这个问题。。。

lsyaizyl, 2020-05-02:
只有使用pwm那个模块就会出现

MrYsLab, 2020-05-02:
What pin number are you using?

MrYsLab, 2020-05-02:
I don't know what you mean by 只有使用pwm那个模块就会出现. Could you please explain.

MrYsLab, 2020-05-02:
Ok I am able to reproduce what you are seeing. I believe that problem is in the s3onegpio code and not s3-extend. I am debugging and will get back to you as soon as I can isolate the bug.

MrYsLab, 2020-05-02:
Fixed with version 1.9. Download the browser files again and it should work. If you are still having issues, leave a comment here and I will reopen the issue.
The problem was that I had a maximum of 1024 for the PWM value which broke the calculations. It should have been 255.

closed 2020-05-02 by MrYsLab

lsyaizyl, 2020-05-03:
非常感谢，我昨晚已经睡觉了，抱歉。没有回复到您

lsyaizyl, 2020-05-03:
[image]

MrYsLab, 2020-05-03:
It should work now with version 1.9 both off-line and online. Just download the files again to use it off line and it should work.

lsyaizyl, 2020-05-07:
大神，请问这个拓展能不能弄到scratch desktop上？？？

lsyaizyl, 2020-05-07:
一定要使用浏览器吗？

MrYsLab, 2020-05-07:
I am not sure if I understand your questions.
If you mean can you use my extensions with the Scratch Desktop that the Scratch team provides, then the answer is no. 

I don't know what you mean by "Are you sure you want to use a browser?". A browser must be used, but if you do not wish to go over the internet, you can [run things locally as explained here.](https://mryslab.github.io/s3-extend/#s3-local/)

I will not be providing a Windows .exe or any other OS executable file,  if that is what you mean.

lsyaizyl, 2020-05-12:
好的，谢谢大神，加油。