MrYsLab/s3onegpio #22: 一但使用pwm的模块就网络中断。。。

lsyaizyl, 2020-05-02:
一但使用pwm的模块就网络中断。。。

MrYsLab, 2020-05-02:
Could you please provide the following information:
1. What type of board are you using (Arduino, Raspberry Pi, etc)?
2. When this happens does s3-extend totally exit?
3. If s3-extend exits, could you please look at the user guide [starting with this link.](https://mryslab.github.io/s3-extend/#ext_ard/)
There is a troubleshooting section for each board type. Follow the instructions and send me any exception output from any of the consoles you will need to open.

lsyaizyl, 2020-05-02:
Raspberry Pi

MrYsLab, 2020-05-02:
Ok. Troubleshooting for the raspberry pi can be [found here.](https://mryslab.github.io/s3-extend/#ext_rpi/#troubleshooting). If there any exceptions printed to the screen for any of the executables, please copy and paste the exception output here.

MrYsLab, 2020-05-02:
Fixed with the resolution of issue #21

closed 2020-05-02 by MrYsLab