MrYsLab/s3onegpio #24: Playground Express LED number

framboise314, 2020-05-09:
Hello Alan
when I command LED number 10 on the Playground Express...
the communication (error message on scratch) is switched off and LED D13 lights up.
My opinion: 
When I turn on LED 1 it's the second LED that lights up.
then up to LED 9 everything is perfect.
but when the script accesses LED 10 it crashes.
can be an offset LED 0 => LED 9
LED 1 => LED 10 ?
thank you
framboise314

MrYsLab, 2020-05-09:
Great catch - thanks. This is fixed in release v1.11. The bug was in the extension and not the Python side and that is why I moved it here. Just start the extension in your browser and it should now work for both off-line and online.

closed 2020-05-09 by MrYsLab

framboise314, 2020-05-10:
Thank you Alan
Sarah and I are completely rewriting the book [Scratch and Raspberry Pi ](https://www.editions-eni.fr/supports-de-cours/livre/scratch-et-raspberry-pi-s-initier-a-l-electronique-et-a-la-robotique-par-le-jeu-9782409011580) for Scratch 3 and Raspberry Pi 4. I'm in charge of Pi4 and extension management. I've already written Arduino and ESP8266, I'm working on Playground Express and I'm testing as many features as possible for chapter writing.  Next step is a real (true) Sparkfun PicoBoard.
The configuration used is a Raspberry Pi 4 with updated Raspbian Buster, and your version of Scratch 3 installed on the Raspberry Pi 4 with the S3 extensions.
For some scripts using sound, it would be interesting to have, in addition to the "When A Loud Sound Is Detected" block that executes a script, a boolean block that does the same thing but that could be integrated to an IF condition.
Thank you for your reactivity, I'll let you have my remarks, you'll see if it should be taken into account.
The book (in French) should come out at the end of the year and it would be a pleasure to offer you a copy. You can reach me directly on contact@framboise314.fr if you want to give me your address :)
ps you can delete this message

MrYsLab, 2020-05-10:
I will respond via email but will leave your message here unless you prefer I delete it.