MrYsLab/s3onegpio #46: Websocket Gateway exited... without pressing ctl+c

headblockhead, 2021-02-24:
The title says it all
[image]

MrYsLab, 2021-02-24:
I am assuming you are trying this with the old Scratch Pico board and not the new Raspberry Pi Pico.
Does this happen immediately after starting Scratch in the browser or after some time?
After you see the error could you please press the F12 key on your keyboard to bring up the browser console (you may have to select a tab to go to the console) and let me know if there are any warnings or errors reported?
Which browser are you using?
Also, could you please try running the troubleshooting steps shown [here?](https://mryslab.github.io/s3-extend/ext_pico/#troubleshooting)

headblockhead, 2021-02-24:
**Does this happen immediately after starting Scratch in the browser or after some time?**

Just around 5 seconds after starting the python script, whether I launch the browser window or not

**After you see the error could you please press the F12 key on your keyboard to bring up the browser console (you may have to select a tab to go to the console) and let me know if there are any warnings or errors reported?**

This is all that shows up, I doubt it is very helpful though.
[image]

**Which browser are you using?**
As you have probably seen from the image above, chrome.

**New development!**
The s3p program now doesn't crash straight away, but instead the picoboard seems to short out as the lights go out when I move the slider too far . This is strange as I have not been able to replicate this on my ubuntu machine running the python code from my repo here: [https://github.com/headblockhead/picoboard](https://github.com/headblockhead/picoboard)

Any help appreciated!

MrYsLab, 2021-02-24:
What versions of Python are you using on your Ubuntu machine and on your Windows machine?
Before you move the slider, does everything stay up? 
If you just use the button and not the slider does it crash as well?
If you could run the troubleshooting scenario (bringing everything up individually) that might help provide some useful information.
Please run it for all scenarios that crash.
When launching from s3p, the exceptions are suppressed, but when if each component is brought up individually, the exceptions should come to each individual terminal window. 
Thanks.

headblockhead, 2021-02-26:
Python 3.9.2 on both. Using windows store version on my windows PC though.
[image]

Everything stays up if I don't move the slider and the button does work.

Could you please elaborate on how I might go about doing that?

MrYsLab, 2021-02-26:
Thanks for providing that information. I have a couple of ideas, but I am unable to test them because I do not own a Pico board. I developed the code for a group in Taiwan a few years back and they did all the testing during development. I do not know if you are using a Sparkfun board or not. It is possible that the Sparkfun board is using different hardware than used in the Taiwan clones. Also, they tested with Python 3.8. Originally, they were using Python 3.7 on Windows, but when Python 3.8 came out and was installed, it broke things. Python changed asyncio in Python 3.8 for Windows. I put some workaround code for Windows to fix the issue. Since you are seeing the issue both on Windows and Ubuntu, most likely Python 3.9 is not the cause, but I can't be sure. I do not run Python 3.9 as of yet, so I have no experience with it.

It appears that perhaps the [polling loop](https://github.com/MrYsLab/s3-extend/blob/ecd94ed211a10aeb9bc440f0464441acc3921b72/s3_extend/gateways/picoboard_gateway.py#L136) used in the pico gateway is running too quickly and flooding things. Adding a short delay may solve the issue. But before I create a branch and make any coding changes, it would be very helpful to find out why the WebSocket gateway is crashing. I am not sure if this will provide any useful clues or not, but without knowing if a useful exception is being thrown when the WebSocket gateway crashes, it is very hard to provide a fix without totally guessing. 

If you could run the troubleshooting scenario, mentioned in a previous posting, when the WebSocket gateway crashes, it should provide an exception traceback in its command window. Again, it may or may not be helpful, but at least it should show where things went wrong by indicating the line of code where the crash occurred. If you copy the traceback and paste it into this thread, I will be happy to analyze what might be going wrong for you. Also, if there is an exception thrown in the other command windows, if you could provide those as well, it would be of great help.

Thanks.

MrYsLab, 2021-03-05:
I have not heard back from you, so I am closing this issue. If you wish to pursue it, please let me know and I can reopen it.

closed 2021-03-05 by MrYsLab

headblockhead, 2021-03-05:
Sorry about that, The email must have been filtered to spam. It turned out that the issue was just with my picoboard. Tried a new one and it worked perfectly! 😀  Thanks for the help though.