MrYsLab/s3onegpio #53: How to read digital input from sensor

pacebrian0, 2021-04-06:
I noticed that the read digital pin block just reads gpio state and not a stream of data.

Is there a way I can read a sensor's data, such as temperature sensor, humidity sensor? They only require one GPIO data pin (apart from power and ground)

MrYsLab, 2021-04-06:
It is possible, and in fact, outside of the Scratch extensions, for Arduino boards, the DHT22 type sensor is supported. The reason that the DHT22 or other temperature/humidity sensors are not supported for Scratch, is that if I support one sensor, inevitably a request would come in for another. Also, streaming type interfaces tend to be complex to support in the context of Scratch.
Currently, the Scratch blocks support the most common digital and analog functionality for the boards supported with some minor exceptions.
What specific sensor are you referring to?

pacebrian0, 2021-04-07:
I am using generic humidity and temperature sensors from Aliexpress. They only require vcc, gnd and out signal. I might be underestimating the work required here, but would it be possible to have a block that reads such a stream of incoming data, or it needs to be defined depending on the type of data being sent from the sensor (i.e. boolean, float, int..)?

MrYsLab, 2021-04-07:
Each sensor type may have a different set of timing requirements. Some require a request for a sample and then require a wait for the data to return. The DHT11 for example requires a wait of 1 second, while the DHT 22 requires a wait of 2 seconds. Accommodating different sensor types would involve providing a block to configure the sensor type, adding additional complexity. It sounds simple but the details of implementation are not. 
I am not sure what a "generic" sensor from Aliexpress is, and so I cannot determine how difficult it would be to implement. If it is a standard part with a standard part number and that adheres to the specs for that part, it may be possible, but without that information, it would be rather difficult.
I am not likely to implement the code for this, but if you would like to try your hand at an implementation, I would be happy to try and provide guidance.

pacebrian0, 2021-04-08:
Thanks for your patience and insight. I will do more research on the matter and get back to you.

Thank you for your fast responses and stay safe

closed 2021-04-08 by pacebrian0

suoko, 2024-02-05:
> Each sensor type may have a different set of timing requirements. Some require a request for a sample and then require a wait for the data to return. The DHT11 for example requires a wait of 1 [...]

Did you write a sort of tutorial on how to implement the support for other sensors or devices ? Let's think of DHT, LCD panels, etc... which might require extra libraries.

MrYsLab, 2024-02-05:
@suoko One must understand how to extend the Scratch side and appropriate hardware support on the board side to add additional support.

My blog has a 5 part posting describing what is involved on the Scratch side. The first part may be [found here.](https://mryslab.github.io/bots-in-pieces/scratch3/gpio/2019/09/15/scratch3-1.html)

To easily find the other articles, please look [at this page.](https://mryslab.github.io/bots-in-pieces/posts/)

On the board side, you must modify the [appropriate gateway](https://github.com/MrYsLab/s3-extend/tree/master/s3_extend/gateways) and its associated board control library. For example, the Arduino AVR boards use [ pymata-express](https://github.com/MrYsLab/pymata-express). 

The gateways are built using yet another package called [python banyan](https://mryslab.github.io/python_banyan/) using the OneGPIO concept.

As you can see, adding functionality is a challenging task. However, if you have any specific questions, I am always happy to try to answer them.