MrYsLab/s3onegpio #63: The current version of S3OneGPIO fails to compile by "yarn start".

KigenHasebe, 2021-05-13:
Hi!

The current version of S3OneGPIO fails to compile by "yarn start".
The failing version of s3onegpio-master.zip was downloaded (Code > Download) on 2021/05/12.
Versions downloaded on 2021/05/08 and 2021/03/19 are OK.

System environment:

Microsoft Windows 10 Pro
10.0.19042 build 19042
node v14.17.0
npm 6.14.13
yarn 1.22.5
git version 2.31.1.windows.1

Build procedureg: 

cd C:\Users\%USERNAME%\s3onegpio-master\scratch-gui
yarn unlink scratch-vm
cd C:\Users\%USERNAME%\s3onegpio-master\scratch-vm
yarn unlink
yarn --force install
yarn link
cd C:\Users\%USERNAME%\s3onegpio-master\scratch-gui
yarn link scratch-vm
yarn --force install
yarn start

Extracted log (after "yarn start"):

yarn run v1.22.5
warning ..\..\package.json: No license field
$ webpack-dev-server
i ｢wds｣: Project is running at http://0.0.0.0:8601/
i ｢wds｣: webpack output is served from /
i ｢wds｣: Content not from webpack is served from C:\Users\kh\s3onegpio-master\scratch-gui\build
Browserslist: caniuse-lite is outdated. Please run:
npx browserslist@latest --update-db
(node:10424) DeprecationWarning: Tapable.apply is deprecated. Call apply on the plugin directly instead
(Use `node --trace-deprecation ...` to show where the warning was created)
× ｢wdm｣: Hash: 92e7e1c762f213e86e09
Version: webpack 4.44.2
Child
    Hash: 92e7e1c762f213e86e09
    Time: 41325ms
    Built at: 2021/05/13 10:49:20
                                                                  Asset        Size                Chunks                         Chunk Names
                                         326c5f98902ccd335b5b.worker.js    6.44 KiB                        [emitted] [immutable]
                                     326c5f98902ccd335b5b.worker.js.map    6.32 KiB                        [emitted] [dev]
                                                       blocks-only.html   456 bytes                        [emitted]    
...
...
                                 static/extensions/example-extension.js    9.76 KiB                        [emitted]
                                                     static/favicon.ico    33.7 KiB                        [emitted]
    Entrypoint gui = lib.min.js lib.min.js.map chunks/gui.js chunks/gui.js.map
    Entrypoint blocksonly = lib.min.js lib.min.js.map chunks/blocksonly.js chunks/blocksonly.js.map
    Entrypoint compatibilitytesting = lib.min.js lib.min.js.map chunks/compatibilitytesting.js chunks/compatibilitytesting.js.map
    Entrypoint player = lib.min.js lib.min.js.map chunks/player.js chunks/player.js.map
     [0] multi (webpack)-dev-server/client?http://0.0.0.0:8601 react react-dom 52 bytes {lib.min} [built]
     [2] multi (webpack)-dev-server/client?http://0.0.0.0:8601 ./src/playground/index.jsx 40 bytes {gui} [built]
    [12] multi (webpack)-dev-server/client?http://0.0.0.0:8601 ./src/playground/blocks-only.jsx 40 bytes {blocksonly} [built]
    [13] multi (webpack)-dev-server/client?http://0.0.0.0:8601 ./src/playground/compatibility-testing.jsx 40 bytes {compatibilitytesting} [built]
    [14] multi (webpack)-dev-server/client?http://0.0.0.0:8601 ./src/playground/player.jsx 40 bytes {player} [built]
     [./node_modules/classnames/index.js] 1.17 KiB {lib.min} [built]
     [./node_modules/core-js/fn/array/includes.js] 109 bytes {lib.min} [built]
     [./node_modules/core-js/fn/promise/finally.js] 166 bytes {lib.min} [built]
     [./node_modules/react-dom/index.js] 1.33 KiB {lib.min} [built]
     [./node_modules/react/index.js] 190 bytes {lib.min} [built]
     [./node_modules/webpack-dev-server/client/index.js?http://0.0.0.0:8601] (webpack)-dev-server/client?http://0.0.0.0:8601 4.29 KiB {lib.min} [built]
     [./src/playground/blocks-only.jsx] 1.09 KiB {blocksonly} [built]
[... 48 lines cut ...]
        Entrypoint main = 326c5f98902ccd335b5b.worker.js 326c5f98902ccd335b5b.worker.js.map
        [./node_modules/babel-loader/lib/index.js?!./node_modules/scratch-storage/src/FetchWorkerTool.worker.js] ./node_modules/babel-loader/lib??ref--4!./node_modules/scratch-storage/src/FetchWorkerTool.worker.js 2.24 KiB {main} [built]
    Child worker:
                          Asset      Size  Chunks                   Chunk Names
            extension-worker.js  55.8 KiB    main  [emitted]        main
        extension-worker.js.map  43.6 KiB    main  [emitted] [dev]  main
        Entrypoint main = extension-worker.js extension-worker.js.map
        [./node_modules/babel-loader/lib/index.js?!./node_modules/scratch-vm/src/extension-support/extension-worker.js] ./node_modules/babel-loader/lib??ref--4!./node_modules/scratch-vm/src/extension-support/extension-worker.js 4.29 KiB {main} [built]
        [./node_modules/scratch-vm/node_modules/minilog/lib/common/filter.js] 1.36 KiB {main} [built]
        [./node_modules/scratch-vm/node_modules/minilog/lib/common/minilog.js] 1.55 KiB {main} [built]
        [./node_modules/scratch-vm/node_modules/minilog/lib/web/array.js] 319 bytes {main} [built]
        [./node_modules/scratch-vm/node_modules/minilog/lib/web/console.js] 925 bytes {main} [built]
        [./node_modules/scratch-vm/node_modules/minilog/lib/web/index.js] 1.28 KiB {main} [built]
        [./node_modules/scratch-vm/node_modules/minilog/lib/web/jquery_simple.js] 2.16 KiB {main} [built]
        [./node_modules/scratch-vm/node_modules/minilog/lib/web/localstorage.js] 530 bytes {main} [built]
        [./node_modules/scratch-vm/src/dispatch/shared-dispatch.js] 14.6 KiB {main} [built]
        [./node_modules/scratch-vm/src/dispatch/worker-dispatch.js] 7.41 KiB {main} [built]
        [./node_modules/scratch-vm/src/extension-support/argument-type.js] 605 bytes {main} [built]
        [./node_modules/scratch-vm/src/extension-support/block-type.js] 1.02 KiB {main} [built]
        [./node_modules/scratch-vm/src/extension-support/target-type.js] 308 bytes {main} [built]
        [./node_modules/scratch-vm/src/util/log.js] 84 bytes {main} [built]
        [./node_modules/webpack/buildin/global.js] (webpack)/buildin/global.js 472 bytes {main} [built]
            + 5 hidden modules
i ｢wdm｣: Failed to compile.
[period]

MrYsLab, 2021-05-13:
Thanks for letting me know. I forgot to check-in the image files for the RPi Pico. This will be resolved in the next release.

MrYsLab, 2021-05-13:
This should now be fixed. Please try downloading the files again and rebuild.

closed 2021-05-13 by MrYsLab

KigenHasebe, 2021-05-14:
Thanks! All are fine now.