MrYsLab/s3onegpio #64: Re: Thanks for adoption of translation for Japanese #43

KigenHasebe, 2021-05-13:
On May 13, 2021, there are no improvements in RPi and Arduino extension.

The problem is confusions in index.js's of those extensions. The conflicts break parameter parts of extension blocks. For example of RPi:

Llines 613~614, 635~637:

    _setLocale () {
        let now_locale = '';
    ...
            case 'ja':
                now_locale='ja';
                break;

Nevertheless, Lines 76~86 etc. have no 'ja'  counterparts:

// common
const FormDigitalWrite = {
    'pt-br': 'Escrever Pino Digital [PIN]como[ON_OFF]',
    'pt': 'Escrever Pino Digital[PIN]como[ON_OFF]',
    'en': 'Write Digital Pin [PIN] [ON_OFF]',
    'fr': 'Mettre la pin numérique[PIN]à[ON_OFF]',
    'zh-tw': '腳位[PIN]數位輸出[ON_OFF]',
    'zh-cn': '脚位[PIN]数位输出[ON_OFF]',
    'pl': 'Ustaw cyfrowy Pin [PIN] na [ON_OFF]',
    'de': 'Setze digitalen Pin [PIN] [ON_OFF]',
};

MrYsLab, 2021-05-13:
I apologize. I am not sure why they were not there, but I have checked all of the extensions and the translations will appear in the next release. I should be releasing within a day or two. I have also added your translations for the Raspberry Pi Pico.
When I release, I will post a message here.

MrYsLab, 2021-05-13:
This should now be fixed for all boards.

closed 2021-05-13 by MrYsLab