MrYsLab/s3onegpio #65: Connection to ESP8266 seems to be unreliable

jstoezel, 2021-05-19:
Hello,

Thank you very much for providing this extension to Scratch3! This is a great initiative.

When working with Scratch3 offline and an ESP8266, I am observing an unreliable connection which makes it almost impossible to exchange data.

I am using Telenetrux-ESP8266 firmware version 1.5 and the currently released version of s3e.
The ESP-8266 connects to the Wifi Network and the TCP server waits for connections. s3e seems to start and run happily.
I can see a connection coming in on the firmware side whenever I start the Scratch3 script ("Client connected to address:...".
In Wireshark I can see packets being exchanged between the ESP8266 and the Scratch3 script. So far, so good.

The first issue happens when I stop the Scratch3 script: the connection to the server does not appear to be detected as closed by the firmware. Even though the script is stopped, the firmware continues to report to the remote client, I can see packets being sent from the ESP-8266 to the client. The firmware only sees a closed connection when I manually stop s3e (ctrl+C). It looks like the "stop_reports" variable is not set on the firmware side, and the connection is not seen as closed as well.

The second issue is that when I stop the scratch script I am unable to communicate with the firmware again even though I start the Scratch script, and reinitialize the connection. I need to kill s3e, start it again, kill Scratch3 and start it again. This seems to be linked to the first issue, where the connection between the client and the server is not closed when the Scartch3 script stops.

Do you have an idea of what the issue might be, and how to fix it?

Regards,
JS

MrYsLab, 2021-05-19:
@jstoezel I have transferred the issue to the s3onegpio repository in case you weren't notified by GitHub.

For the first issue, Scratch3 does not provide any external notification for the extension to use when the user hits the Stop button. In Scratch2 they did provide this facility, but for some curious reason removed it in Scratch3. I have documented the "approved" method of shutdown [here.](https://mryslab.github.io/s3-extend/shutdown/)

For the second issue, I am not able to reproduce this in either Windows or Ubuntu Linux.
I am able to start and start the Scratch script below multiple times without issue.
[image]

If you can provide a Scratch script that demonstrates the issue, that would be of great help.

Thanks.

MrYsLab, 2021-05-23:
May I close this issue or do you still need some help?
Thanks.

MrYsLab, 2021-05-25:
I am closing this issue because I did not receive a reply to my previous posts., If you feel it still needs to be open, please leave a comment here and I will reopen it.

closed 2021-05-25 by MrYsLab