MrYsLab/s3onegpio #85: Request to host these extensions in my own Scratch 3 fork

dalelane, 2022-01-19:
Hello! 👋

I run a fork of Scratch 3 at https://machinelearningforkids.co.uk/scratch3/ which is Scratch + some extensions I've written to allow use of machine learning models. 

I've had a request from a user who wants to use both of our extensions in a single project - an AI/ML project that controls an Arduino or Raspberry Pi. 

Would you be happy for me to add a copy of your extensions from this repo into my fork as well? 

(If you have any hesitation or concerns about this at all, please just say - and I wouldn't do it)

MrYsLab, 2022-01-19:
Hi Dale,
    I would be pleased if you added my extensions to your project. If you have any questions, please feel free to ask.
Alan

dalelane, 2022-01-19:
That's very kind - thank you!

closed 2022-01-19 by dalelane