MrYsLab/s3onegpio #98: Install s3onegpio extension in fresh Scratch does not work

SebCanet, 2022-08-28:
Hell Alan,
thanks a lot for your work! So I wanted to use your documentation to add my extension in Scratch3 and it works, thanks a lot for your link to hiroyuki tuto. But if I copy needed files for s3onegpioArduino, for example, I have error in browser (Firefox 104, Ubuntu 64 bits) and maybe some code has changed in new version of Scratch.
It compiles with webpack and I can launch Scratch3, but when opening extension panel I have error "[React Intl] Missing message: "gui.extension.scratchhome.description" for locale: "fr", using default message as fallback.".
If I continue with Scratch in english of course this problem disappears, but when I select ongpioArduino extension in console I have this error:
[image]

Can you updates scratch-gui & scratch-vm please ?
Many thanks!

SebCanet, 2022-08-28:
New test: in first case I compiled then added Arduino extension, but if I l delete node_modules and launch again "yarn install", the "yarn start" does not compile:
ERROR in ./node_modules/scratch-vm/src/extensions/scratch3_onegpioArduino/index.js

    Module not found: Error: Can't resolve 'sweetalert' in '/home/babas/Téléchargements/Scratch3_git_sansMaJ/scratch-gui/node_modules/scratch-vm/src/extensions/scratch3_onegpioArduino'
     @ ./node_modules/scratch-vm/src/extensions/scratch3_onegpioArduino/index.js 44:0-21
     @ ./node_modules/scratch-vm/src/extension-support/extension-manager.js
     @ ./node_modules/scratch-vm/src/virtual-machine.js
     @ ./node_modules/scratch-vm/src/index.js
     @ ./src/containers/controls.jsx
     @ ./src/playground/blocks-only.jsx

    ERROR in unable to locate '/home/babas/Téléchargements/Scratch3_git_sansMaJ/scratch-gui/node_modules/scratch-vm/dist/web/extension-worker.{js,js.map}' glob

Still trying...

SebCanet, 2022-08-28:
After added sweetalert in Scratch-vm, now scratch-gui warns: "ERROR in unable to locate '/home/babas/Téléchargements/Scratch3_git_sansMaJ/scratch-gui/node_modules/scratch-vm/dist/web/extension-worker.{js,js.map}' glob"

MrYsLab, 2022-08-28:
@SebCanet Hi Seb, 
I just rebuilt using a VirtualBox install of Ubuntu 22.04, and s3onepgpio is working for me. Here are the steps I took:
1. I installed node.js version v12.22.9 using the instructions from [this page](https://computingforgeeks.com/how-to-install-nodejs-on-ubuntu-debian-linux-mint/).
2. I installed npm using 
```
sudo apt install npm
```
3. I installed yarn with:
```
npm install yarn
```
4. I cloned s3onegpio:
```
git clone https://github.com/MrYsLab/s3onegpio.git
```
5. I followed [these instructions](https://github.com/MrYsLab/s3onegpio/blob/master/notes/rebuilding_and_rerunning.txt) to do the build.

6. I opened firefox to  http://0.0.0.0:8601/

If you do the same, does it still fail? I am assuming you had not made any changes to the code when you did a yarn start.

SebCanet, 2022-08-28:
Sorry, I wasn't clear, in fact from a scratch-gui + scratch-vm clone from the official Scratch repository (and thus the latest versions) I tried to insert the s3onegpio extensions.
The manipulations you mention work very well, the problem is not s3onegpio, it's rather how to add the extensions one by one in a recent installation of Scratch. Or maybe is it simplier to update scratch in your s3onegpio repo?
Thanks in advance.

MrYsLab, 2022-08-28:
The "official" Scratch code is a mess. It is probably easier to start with my code based on an older version of Scratch simply because it works. I have not kept up with the latest Scratch changes. I tried that in the past, and it took me many hours to make things work again with no added benefit. The Scratch team does a miserable job documenting what their updates "fix."
Besides getting the latest version, is there anything missing from Scratch using my code as a basis?

SebCanet, 2022-08-28:
Ok, it seems that for many years, collaborating with Scratch team seems impossible. So I'll try to change my extension (it communicates with building in SweetHome3D and I would like to add it with board communication to help kids create a toolchain from design to test on a scale model, with same sensors and actuators than in 3D model) to make compatible with your version.

Between this 2 version of Scratch, this is what I noticed only that there is no more backpack, except that everything seems the same. So the work is maybe in 'back office'.

MrYsLab, 2022-08-28:
No problem. If you have any questions about creating an extension, I will try to answer them if possible. There are so many files to touch when adding an extension. 

I, too, am not very impressed with the Scratch team's support. The reason is probably due to the team consisting of an ever-changing group of graduate students. Also, since it's MIT, they think they know more than anyone else ;-). I will close this issue, but if you comment here, I will see the comment.

closed 2022-08-28 by MrYsLab