MrYsLab/s3onegpio #100: Firefox not able to open project and do not remember installed extensions.

Marooned-MB, 2023-09-09:
Not sure if this project is still maintained but I wanted to report that when working on Firefox, I didn't find a way to open project from file. It just do nothing. No errors in JS console, nothing. It works however in Edge (I don't have Chrome to test).
It means every time I start the page I have to recreate project from scratch (pun not intended).
Tested on off line version (from this repository) and on online version https://mryslab.github.io/s3onegpio/

Firefox: 117
Windows 10

MrYsLab, 2023-09-09:
Yes, the program is being maintained. You didn't mention which board you loaded, but I tested the online version and loaded the Arduino extension.

I created the following simple script and clicked File/Save To Computer.
[image]

I then restarted the browser, clicked on File/Load From Your Computer, and selected the file saved above.
[image]
 
This loaded the script, and it was able to run.

Does this work for you? If you need help, please let me know which board you use so I can test it here.

Marooned-MB, 2023-09-09:
Ah, yes. I use ESP8266 board (Wemos D1 mini).
It also disconnects randomly and I have hard time to reconnect it again. No errors, no info, just stops responding (especially when using servo but I'm not sure yet if only then or it was just coincidence).

As for saving/loading - no, I've tried a few times, saved a few projects to few files, no luck with loading them.

Thanks for fast response.

MrYsLab, 2023-09-09:
I just tried with the ESP-8266 extension, and I can save and restore a script. Try pressing the F12 key after opening Firefox. Select the console tab and then continue as normal. This may show some errors; if so, please copy and paste them here.

Do you also lose connectivity with Edge? 

On my setup. The ESP8266 has problems communicating with my router when an IP address is assigned. I have to reset my router. This only happens with the ESP8266 and no other device. 
Perhaps you are having a similar problem. To check this, the assigned IP address should be printed if you bring up the Arduino IDE Serial Monitor, Power up the ESP8266, and enable the Arduino Serial Monitor. If communication fails, a long series of dots appears in the serial. Monitor window. The IP address can take up to 15 seconds to be assigned.

Marooned-MB, 2023-09-09:
As stated in my first post, no JS errors in the console.
As for the Edge, I've spent a bit playing around and it works for a bit (like a minute or so) then I get popup that websocket disconnect and then I have to try different approaches to reconnect it again involving:
1. resetting uC
2. refreshing webpage
3. shut down and re-run `s3e`
No idea which point does the trick or what combination. Maybe would be a good idea to show some kind of indicator that websocket is connected and separate indicator that connection to uC is established?

So Edge at least can load my project from file.

MrYsLab, 2023-09-09:
I am trying to understand why Firefox is causing your issue. I am using the same version as you. 

To debug the connectivity issue, follow the Troubleshooting section of [this link](https://mryslab.github.io/s3-extend/ext_esp/). 
This will help isolate if the problem is with WebSocket connectivity or in the ESP8266 gateway.

I did find this quote on the [WebSockets page](https://websockets.readthedocs.io/en/stable/topics/timeouts.html):

> If your browser-based app mysteriously and randomly fails to receive events, this is a likely cause. You need a keepalive mechanism in the browser to avoid this scenario.

However, I don't know if this is the cause of the problem, I will need to try and reproduce things here before trying to implement a watchdog timer mechanism.

What processor is your computer using? If it is an older or slower processor, it may contribute to the issue. I am not suggesting a new computer, but it will help me to know where to look.

Marooned-MB, 2023-09-09:
I'm currently testing it on the notebook with CPU i5-6200 @ 2,3GHz and 16GB of RAM.

After I reset all components it seems to work, then I change something in the sketch and bam, it stops responding and even restarting all 3 components (browser, uC and s3e) it's not coming back. I have to restart it multiple times in various order to get it working... for a short time. All those tests were done on Edge right now to exclude Firefox issues.

Anyway I have to finish for now. Thanks for your your help! Much appreciated!
Not sure if I'll find another time slot for testing but I found it too unstable to play with it more (I was planing to give it to my 8yo daughter who was coding in regular Scratch just next to me - wanted her to be able to play with some physical devices but looks like we'll stay with original Scratch for now :) )

Thanks

MrYsLab, 2023-09-09:
Ok. Thanks for the info. Your laptop is certainly fast enough. I will be testing on my end when I get a chance. If I find anything, I will let you know.

MrYsLab, 2023-09-09:
I played some more, and I do not see the disconnecting problem. The cause might be not having an adequate power supply for the ESP8266. 

I am going to close this issue since I cannot reproduce it. You may still leave a comment here, and I will see it.

closed 2023-09-09 by MrYsLab