MrYsLab/s3onegpio #104: s3rp crash on latest Win10/ Google Chrome when using read digital pin

ssoltek, 2023-12-30:
Actually doing some first steps with a Raspberry Pico and your great scratch one GPIO extension. Settings digital pins an doing blinking for example works fine, but using read digital pin leads to immediate cloding of the websocketServer because s3rp exits with signal 0:

PS C:\Users\Lucy> s3rp
backplane started
Websocket Gateway started
RPi Pico Gateway started
To exit this program, press Control-c
Websocket Gateway exited...
Exception ignored in atexit callback: <bound method S3RP.killall of <s3_extend.s3rp.S3RP object at 0x0000017408FC4230>>
Traceback (most recent call last):
  File "C:\Users\Lucy\AppData\Local\Programs\Python\Python312\Lib\site-packages\s3_extend\s3rp.py", line 141, in killall
    sys.exit(0)
SystemExit: 0
PS C:\Users\Lucy>

See screenshot for actual program.
[image]

 python, pip and s3extend are up to date. What i find curiuos is that the screenshot in your manual is not matching the acual scratch command - the option to configure the pullup resistor is missing. May this be the problem?

Anything I can do beeing a python noop to assist in debugging?

Bye and a happy new year
Simon

MrYsLab, 2023-12-30:
Hi Simon,
    The problem should be fixed. Please let me know if it works for you so I can close the issue.
    The parameter should now appear in the block.
    Also, I'd like to point out that I moved the issue to the s3onegpio repository since that is where I implemented the fix.

   Thanks again for reporting the issue.

ssoltek, 2023-12-31:
Thx!
Fix is working fine for me (actually tested on Linux today because having actually no access to the windows machine)

Happy new year!

MrYsLab, 2023-12-31:
Great. Have a very Happy New Year.

closed 2023-12-31 by MrYsLab