MrYsLab/s3onegpio #105: Translation for Dutch (NL)

tivanboven, 2024-02-28:
[playground_express.txt](https://github.com/MrYsLab/s3onegpio/files/14431275/playground_express.txt)

MrYsLab, 2024-02-28:
Thank you for providing the Dutch translations. Implementing this for all the extensions may take me a couple of days. I'll let you know when it's finished and ready to use.

MrYsLab, 2024-02-28:
I have implemented your changes for the Playground Express, but they still need to be published.  
For the Arduino, ESP-8266, Raspberry Pi, Raspberry Pi Pico, Picoboard, and RoboHat, the blocks will appear in English if the user selects Nederlands in Scratch.

If you wish to provide translations for the Arduino, ESP-8266, Raspberry Pi, and Raspberry Pi Pico, please update the [translations.txt file. ](https://github.com/MrYsLab/s3onegpio/blob/master/translations/translations.txt) This is optional on your part.

The Picoboard is out of production, so providing translations for that board seems unnecessary. The RobotHat is being used sparingly, but if you wish to provide translations, I can provide a file with its current block translations.

Again, this is totally up to you. I am not trying to add more work for you.

Please let me know if you wish to provide any additional translations. If you do, I will wait until you can provide them, and if not, I will publish the changes for the Playground Express.

tivanboven, 2024-03-04:
> I released the changes for the Playground Express and updated the user
> guide, giving you credit for the translation.
>
> Thanks,
> Alan

Hi Alan,

Thanks for your quick turnaround! I haven't prioritized translating the
other files yet. Though I am wondering: wouldn't generative ai be able to
do pretty good translations for these? Maybe we can check for the languages
we know and if it's accurate, assume it will be accurate for other
languages too?

Cheers,
Thijs

tivanboven, 2024-03-04:
I've done the translation for the file you linked, see attached. I didn't go through the trouble of trying GenAI to do this though, so no conclusions there.
[translations.txt](https://github.com/MrYsLab/s3onegpio/files/14485242/translations.txt)

MrYsLab, 2024-03-05:
Thanks. I updated the extensions using your translations. All should now be available.