MrYsLab/s3onegpio #106: ERROR: After building locally not able to load extensions locally

codezuma, 2024-11-08:
[image]

Getting this error while clicking on ardino extension

MrYsLab, 2024-11-08:
I cannot reproduce the error you reported. Building can be somewhat tricky. I build in Ubunu Linux using within the [WebStorm IDE](https://www.jetbrains.com/webstorm/). I don't know if building in another operating system or IDE works or not.

I use:
    yarn 1.22.19
    node v19.9.0

I select the version of node using the[ nvm package manager.](https://github.com/nvm-sh/nvm)

Within a terminal window I issue the following commands:

nvm use  19.9.0
export NODE_OPTIONS=--openssl-legacy-provider

cd to scratch-gui
yarn unlink scratch-vm
cd to scratch-vm
yarn unlink
yarn --force install
yarn link
cd to scratch-gui
yarn link scratch-vm
yarn --force install
yarn start

If your build using the tools and environment specified above, do you still see the error?
Are you running your browser using the link provided by yarn start command?

MrYsLab, 2024-11-08:
If you can successfully build and then run within _yarn start_, to create a standalone copy you will need to do build and deploy to your fork of this repository.

To do this, cd into scratch-gui, and then run the following two commands:

yarn run build
yarn run deploy

This will upload the code to your forked repository in the gh-pages branch.
Follow [these instructions ](https://mryslab.github.io/s3-extend/s3-local/)to run locally.

codezuma, 2024-11-09:
Looks like there was a problem with yarn version I swtiched to the one you provided and it got resolved. @MrYsLab  thanks a lot.

closed 2024-11-09 by codezuma